MegaPirateNG/FlashTool #22: Failed to Download Firmware Inormations

bpopp, 2015-01-13:
When trying to run the flashtool, I get the error, "Failed to Download Firmware Informations". Occasionally it will start to download, but never gets past 1%. I looked through the code, and it's apparently trying to pull down the XML file from "http://fw.megapirateng.com:8888/update.xml" which seems to be offline.

SirAlex, 2015-01-13:
Hi, thanks for report, please try again. Build server was down for some
reason.

bpopp, 2015-01-13:
Yep, that did it. Thanks!

closed 2015-01-13 by bpopp

eltesco1, 2015-05-12:
Hi sir Alex I am having this same issue again, also cannot browse to megapirateng.com seems site may be Down?

SirAlex, 2015-05-13:
Sorry, but server is down due out of money to pay for colocation service at DataCenter.

eltesco1, 2015-05-13:
Hi Sir Alex

I am getting this error on compile:
 avr-gcc error:unrecognized command line option'assembler-with-cpp

*But I cannot find instructions to resolve, I think they are on the
megapirate forum which is Down?*

*Do you know how to resolve?*

*Thanks for your help*

AlaskaJ, 2015-05-15:
Thanks SirAlex for all your work. For a non-techie your flashtool is a lifesaver cause I can't compile a sketch to save the life of me. Is there any way to have the flashtool work without going to the server by instead using firmware locally and the user just go to the site to update them?

etechnox, 2016-08-09:
Now the megapirateng.com had close , is there anywhere I ccanget the documents and firmwares? The flash tools I downloaded can't work as it try to access the site.

smurfy, 2016-08-09:
i could host another build server if its required and still used.

SirAlex, 2016-08-09:
Build server still running, but domain name was not prolonged.

smurfy, 2016-08-09:
can you provide an alternate domain? i

etechnox, 2016-08-09:
I just bought another new crius aiop , just know that the website already close ... I can't really find the method to install the firmware from internet , maybe the megapirateng documentation is still useful, hope the domain still exist

smurfy, 2016-08-09:
the docs are served by github with a cname to docs.megapirate....
I will remove that and edit this post with the "new" link.

etechnox, 2016-08-09:
Can I download the firmware also , the flashtool can't use right? I had
mistakenly update with mission planner on the attempting process

smurfy, 2016-08-09:
Currently the flashtool is unusable, because it can't reach the build server.
But you can build it yourself as described in the docs.

etechnox, 2016-08-09:
I tried the who night , this error keep come out
avr-gcc: error: unrecognized command line option '-assembler-with-cpp'

danil2009, 2016-08-13:
Hi Sir Alex
I want to back My quadcopter firmware using MegapirateNg 3, using flashtools. But now i cannot use flashtools. I hope you can fix the server.
thanks
Dida

SirAlex, 2016-08-15:
@smurfy 
>can you provide an alternate domain?

Please, try to use **mpng.siralex.info** instead of **fw.megapirateng.com**

smurfy, 2016-08-15:
@SirAlex will do another release soon. you need to change the url in the http://mpng.siralex.info:8888/update.xml as-well

SirAlex, 2016-08-15:
@smurfy 
>http://mpng.siralex.info:8888/update.xml as-well

Done. :)

meggerja, 2016-08-17:
Hello, can you tell me is there any normal way to use this flash tool ? I am not a IT gay like you and I have no idea how to make this working.  Thanks

dzmitryDubrovin, 2016-08-18:
@meggerja 

For Windows:
at start menu type "Notepad" -> MouseRightClick at "notepad" menu item and select "Run as Administrator"
then at Notepad go to File -> Open, at opened window go to C:\Windows\System32\drivers\etc and open file "hosts" 
(**NOTE** that Notepad basicaly tries to open *.txt files and "hosts" doesn't have txt ext, so select "All files" at right corner dropdown)

add following at NEW LINE at end of file  
212.98.191.17 fw.megapirateng.com

save it and rerun FlashTool

smurfy, 2016-08-18:
I will release a new version soon. @dzmitryDubrovin  solution should work too.

meggerja, 2016-08-18:
Thank you ,  that's working :)

smurfy, 2016-08-18:
Also i have just released a new build: https://github.com/MegaPirateNG/FlashTool/releases

@SirAlex please also change the current version in the update.xml to 1.1r3

SirAlex, 2016-08-19:
@smurfy 
Done, thanks for new version :)