MegaPirateNG/FlashTool #25: Server Not Found

hunterway55, 2015-05-21:
Hey, When flash tool trys to update software it reply's with failed to download firmware information. I tried hitting the url and its down. http://fw.megapirateng.com/update.xml http://fw.megapirateng.com:8888/update.xml. Is their a fix for this? Or an expect time that the server will be back up?

closed 2016-08-16 by smurfy

ATi7, 2018-11-07:
Flash tool work  at this date ? I can use it ! ...what are the steps to configure and use it I understand that the server does not work....

smurfy, 2018-11-07:
It works but without server its tricky. Ive updated the project for easier local usage.

ATi7, 2018-11-08:
Thank you to respond me friend very happy ....
Could you explain me  in an easy way how to do it or could you tell me the steps?

smurfy, 2018-11-08:
take a look at the https://github.com/MegaPirateNG/FlashTool/blob/master/README.md

ATi7, 2018-11-08:
If I had already read that ... I am not really a programmer, but I try ,I see QT  consumes memory in my PC ... but I am going to try to see where I arrive.....

ATi7, 2018-11-08:
smurfy ...please I finish download the Qt 5.1 ...

what is the next step or or what do I have to do ..considering that I'm not a programmer...
But I would appreciate if you could guide me...

ATi7, 2018-11-08:
you can enumerate the steps with a  explanation ?

ATi7, 2018-11-09:
Please where update the proyect to local used ?

SirAlex, 2018-11-09:
@ATi7 
Please, try again to use FlashTool, server now works fine.

ATi7, 2018-11-09:
Thank you very much ... I want to thank smurfy and Sr Alex for their help and for their explanations ... thank you for your excellent work and make it possible for us to use your information to be able to do our projects ... very happy for me and  God bless you and all ...

ATi7, 2018-11-09:
Thank you smurfy and Sr Alex ..OK fine ... Sr Alex ....Is possible you have the motor map and the receiver  connection from  hexa frame  in crius aio pro v2 from megapirate ng 3.01r4 and 3.01.5r2 software ? I try see in the git hub  but the link not work

ATi7, 2018-11-10:
Only I need the motor maps from 3.01r4 and 3.01.5r2 ....the motor maps link in the code not work ...Do you have available this ? please ...

ATi7, 2018-11-10:
Only I need the motor maps hexa frame from 3.01r4 and 3.01.5r2 ....the motor maps link in the code not work ...Do you have available this ? please ...

ATi7, 2018-11-14:
I can not  see the motor map from code 3.01r4 to hexa frame  .....Can I get it ? only I need this to finish my proyect ...

ATi7, 2020-05-05:
Hello A Greeting forl all .......I report today the megapirate ng  flash tool not work ....please you can help me ? 
I would like to report a problem that I used the  3.01r4 code and that is that when all  is turned off on in a crius aio pro   ...this  activated and starts the motors sometime ...First I calibrate all motor and I used esc simonk in 1060 to 1860 calibration in my radio and the radio is futaba 9c ....
my question is would it be some program error or my error ....You can help me .....