MegaPirateNG/FlashTool #26: Fehler bei der Aktualisierung, bitte später erneut versuchen

manfred59, 2015-06-03:
Wird das wieder gefixt? Würde gerne mit Flashtool meinen Crius wieder zum Leben erwecken ;))

goial, 2015-06-05:
Ich wollte eben von 3.0.1 auf 3.1.5 wechseln.. Eeprom clear -> flashtool starten -> server offline :( und es ist so schones wetter!!! :((( hätte mal erst gucken sollen ob das flashtool funktioniert :O

smurfy, 2015-06-06:
@goial it makes no sense compiling the flashtool without said portnumber :) while the update.xml maybe available on port 80 too, the buildserver itself is not. it's a separate server running via node js on said port 8888.

goial, 2015-06-06:
ok, that make sense :)
The FlashTool is so comfortable to use.. Do you know whats the Problem is? Server Crash or not enough money? I'd like to help if i can :)

SirAlex, 2015-06-06:
Flashtool must work now.

@smurfy 
Did you know how to start "forever" server (node js)  when server startups? Looks like server sometime rebooting (powering issue), and I must login to it and manually start scripts.

SirAlex, 2015-06-06:
@smurfy 
Nevermind, followed this guide http://stackoverflow.com/questions/13385029/automatically-start-forever-node-on-system-restart , now it will start automatically.

goial, 2015-06-06:
@SirAlex Thank You it works fine!!!! You are the best! :))))

closed 2015-06-07 by smurfy