MegaPirateNG/FlashTool #27: Is the build server working?

Espenf, 2016-12-30:
Hi, 
Would it be possible to have an compiled working version?
When i try to run the exe file it ask for a lot of Qt-dlls.  I also copied these to folder but it does not help.
Compile gives some warnings like these:
C:\Qt\Examples\FlashTool-1.1r3\client-side\F4BYFirmwareUploader.cc:70: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
     for (unsigned i = 0; i < src.size(); i++)
                                     ^
I have tested with several QT 5.1 mingw32-make.exe and upwards..

Is the build server working? 
http://mpng.siralex.info:8888/hex%22%20flashToolVersion=%221.1r3%22 => 
Cannot GET /hex%22%20flashToolVersion=%221.1r3%22

Would it be possible to have an compiled working version?

Regards Espen

SirAlex, 2017-01-03:
Hi, Espenf! Happy NY!

Why you trying to compile FlashTool? Build server working fine.

Espenf, 2017-01-03:
Hi Alexey and Happy New Year to you and yours!
If I dont have to do create it I will be happy. 
Please send me zip or rar containing an .exe, I don’t have it anymore and by downloading here no .exe is included.
Regards Espen

smurfy, 2017-01-03:
Hi,

You can download it here: https://github.com/MegaPirateNG/FlashTool/releases
Documentation can be found here: https://megapirateng.github.io/Docs/documentation/flashtool

Espenf, 2017-01-04:
Thanks PhA and Happy New Year to you also!  I was so stupid not finding the release tab :)

closed 2017-01-04 by Espenf